nasa/astrobee #794: No CMake list in home directory

Jvogie, 2024-06-12:
Hello, I am trying to add a sub directory with some path planning algorithms that I would like to run and followed the "Getting Started" guide provided in the repo. However, it seems that I dont have a CMake list in my home directory but I can run the simulation and use the tele op tool just fine. Is there a different way I should be adding a new sub directory so that I can run it in the simulation environment? Thanks for the help in advance.

Jvogie, 2024-06-12:
Yes sorry for the confusion. In "A Brief Guide To Astrobee v 1.1" section 7.1 it says "Packages can be added in any directory in the $SOURCE DIR, e.g., freeflyer-shared/$MY NEW PACKAGE DIR. New directories can be added to the existing file structure of the source code as well, but the CMakeLists.txt sequence must be modified to find them." So going off of that I added my Algorithms folder to my source directory which is titled "astrobee". Then in section 7.2 it says "Assuming a new directory has been added, modifying the main CMakeLists.txt,
located in $SOURCE DIRECTORY would be appropriate to include the $MY NEW PACKAGE DIR directory.". However when I go into the source directory I have no CMakeLists.txt only the build, catkin_tools, devel, log, src folders and the newly added Algorithms folder. 

I can run the astrobee simulation environment and see the astrobee robot but if I call a command to run one of my programs I get the error "[rospack] Error: package 'execute_asap' not found". Where 'execute_asap' is my coordinating script. This algorithms folder I have was given to me by a group who has done other testing with it so I believe the problem is that astrobee just isn't recognizing its there.

Let me know if that clarifies things or if you need any other information to understand my problem. Thanks again!

bcoltin, 2024-06-13:
Hi @Jvogie , we changed our build system a few years ago and I believe the guide may be out of date in this aspect. We now use the standard catkin build system. Each package has its own CMakeLists.txt and there is no top level one. All of the packages need to go in the src folder which it sounds like may be the problem you are having. You can learn more about catkin here: http://wiki.ros.org/catkin/Tutorials

closed 2024-07-08 by bcoltin